IRNAS/koruza-compute-module #3: Ethernet connector

VojislavM, 2017-01-19:
The Ethernet connector is:
* http://www.digikey.com/product-detail/en/bel-fuse-inc/0826-1L1T-57-F/507-1431-ND/2107996

Need to check:
* Ethernet PoE pins
* Ethernet pins 5 and 10, do they go to GND?
* Ethernet does shield go to GND?

[image]

VojislavM, 2017-01-23:
This connector can be also find at mouser electronics: http://eu.mouser.com/search/ProductDetail.aspx?R=0virtualkey0virtualkey0826-1L1T-57-F

VojislavM, 2017-01-24:
Pins 4 and 5 coming form the cable side are positive end of passive PoE power supply, and pins 7 and 8 are negative end (GND).

Pins 5 and 10 on the connector side are not used, and they will be connected to the ground over zero resistors

VojislavM, 2017-08-16:
the connector for the new version has same configuration, but instead of connecting pins 5 and 10 to the ground they will be also used for power. This is the case with gigabit PoE. So eather passive PoE or gigabit PoE is connected the module will have power.

VojislavM, 2017-08-17:
[image]

SloMusti, 2017-10-07:
Gigabit power checked with Ubiquiti Gigabit PoE supply and works as expected.