IRNAS/koruza-compute-module #8: I2C-0 Bus

VojislavM, 2017-01-30:
## I2C-0
As part of the B+ improvemets, the Raspberry Pi Foundation has standardized the interface to add-on boards, in what they call the “Hardware Added On Top” (HAT) specification. It standardizes the physical form factor for add-on boards, and includes a provision for the B+ to automatically identify and initialize HATs at startup. It uses an I2C bus to read a description from an EEPROM on the HAT, similar to cape identification on the Beagle Bone Black.

This capability has been carried forward on the A+ and Pi 2 Model B as well. This I2C bus is found on the ID_SC and ID_SD pins (pins 27 and 28 of the 40-pin connector) - but before you get too excited about adding peripherals on that bus, observe the note in the schematic for that port.
[image]

This is further clarified in the [HAT design guide](https://github.com/raspberrypi/hats/blob/master/designguide.md):

> On a Model B+, GPIO0 (ID_SD) and GPIO1 (ID_SC) will be switched to ALT0 (I2C-0) mode and probed for an EEPROM. These pins will revert to inputs once the probe sequence has completed.
The only allowed connections to the ID_ pins are an ID EEPROM plus 3.9K pull up resistors. Do not connect anything else to these pins!

It’s only there to talk to EEPROMs at addresses 0x50 during boot time. User access at runtime is problematic. If you want a general purpose I2C bus on the B+, you’ll need to use I2C-1, on pins 3 and 5 of the 40-pin connector, marked SDA and SCL on the Pi Wedge.

### Enabling I2C-0
I2C-0 is disabled by default. To enable it, you’ll need to manually edit the configuration file.

Edit /boot/config.txt, and add the following line. If you previously used raspi-config to enable I2C-1 and SPI, you’ll see similar entries near the bottom of the vile.

```
dtparam=i2c_vc=on
```
With that enabled, restart your Pi (sudo reboot). When it’s back up, you’ll know it’s been activated is you’ve got a filesystem node at /dev/i2c-0.

Source: [Link1](https://learn.sparkfun.com/tutorials/raspberry-pi-spi-and-i2c-tutorial)

VojislavM, 2017-01-31:
## I2C 0 interface
It is used for Camera 1, and it cannot be used for other communication see the #4 .

## I2C 1 interface
This port will be used to control two SFP modules. Problem is that both modules have the same address, so they cannot be on the same bus pins. 
####Options:
* use two different pins for the same I2C-1 interface, GPIO2/3 and GPIO44/45. Then the Rpi CM will do the runtime switching in the SoC GPIO multiplexer, and simulate two I2C ports for the SFP modules.
* if runtime GPIO switching is not possible one pin pair will be used for I2C communication and additional GPIO pin will be used to drive external I2C MUX or regular GPIO MUX between two SFP connectors.
* use bit-banging I2C on second GPIO pin pair.

kostko, 2017-02-01:
We can switch the GPIO pin functions during runtime, using the [`wiringpi` utility/library](http://wiringpi.com/the-gpio-utility), which is available in the package repository and may be installed via apt-get. For example:
```
gpio -g mode 2 off
gpio -g mode 3 off
gpio -g mode 44 alt2
gpio -g mode 45 alt2
```

There is also a library, so we can also switch functions from withing the `koruza-driver` daemon.