IRNAS/koruza-compute-module #29: Add LED disable transistor

SloMusti, 2017-08-09:
Add transistor such that all LEDs inside can be disabled, except ethernet port diodes. This prevents from extra light exitign the unit when not needed.

VojislavM, 2017-08-17:
the diodes will be switched with one [MOSFET](http://uk.farnell.com/nxp/nx7002bk/mosfet-n-ch-60v-0-27a-sot-23-3/dp/2469640?ost=2469640&searchView=table&iscrfnonsku=false&ddkey=http%3Aen-GB%2FElement14_United_Kingdom%2Fsearch)

VojislavM, 2017-08-23:
there are two diodes and they are turned off with separate mosfets Q4 and Q5, but with the same pin. Default state is on which is provided by the pull up resistor R64. 

[image]

closed 2017-09-12 by VojislavM